khoih-prog/TimerInterrupt_Generic #1: Support for STM32/STM32Duino?

GitMoDu, 2020-11-11:
Since you're targeting Arduino support, STM32Duino is very popular nowadays.

Library looks great btw.

khoih-prog, 2020-11-11:
Thanks for your encouraging words and using the library.

In the rush, I just somehow **forgot**  to include the [**STM32_TimerInterrupt**](https://github.com/khoih-prog/STM32_TimerInterrupt), which is currently supporting only ststm32,  into this [**TimerInterrupt_Generic Library**](https://github.com/khoih-prog/TimerInterrupt_Generic).

Will add support to **STM32Duino** into [**STM32_TimerInterrupt**](https://github.com/khoih-prog/STM32_TimerInterrupt), then add to this [**TimerInterrupt_Generic Library**](https://github.com/khoih-prog/TimerInterrupt_Generic) within days.

Best Regards,

khoih-prog, 2020-11-13:
Hi,

Just update to add [**STM32_TimerInterrupt Library**](https://github.com/khoih-prog/STM32_TimerInterrupt) to the mess ;-)

---

### Releases v1.2.0

1. Add **STM32_TimerInterrupt** Library

Cheers,

closed 2020-11-13 by khoih-prog