khoih-prog/TimerInterrupt_Generic #3: Multiple Defintions error

Szum123321, 2021-04-22:
Hi!

I'm trying to compile some code for Teensy 3.5 with PlatformIO  2.3.2, and I'm getting multiple definitions error.

I followed the procedure described in README, which sadly didn't work even though I cleaned the build environment and restarted VScode all the time.

Error message looks like this: [link](https://pastebin.com/1ityGyLZ)

I'm using Arch based linux distro:
Linux sz-81nh 5.11.14-arch1-1 #1 SMP PREEMPT Wed, 14 Apr 2021 12:06:34 +0000 x86_64 GNU/Linux

Also, the same code (the only difference being instead of TeensyTimer, I used STM32Timer) compiles and works perfectly on ST Nulceo F303RE

### Steps to Reproduce
Minimal example: [Teensy_minimal.zip](https://github.com/khoih-prog/TimerInterrupt_Generic/files/6360680/Teensy_minimal.zip)

### Expected behavior

Code compiles for Teensy

### Actual behavior
Linker spews out a bunch of `Multiple definitions error`

khoih-prog, 2021-04-22:
I don't see you use this library anywhere in the code.

It seems that you're using `TeensyTimers` library and there is something wrong with it and PIO.

I'm closing the issue now as invalid, until you prove that the issue is caused by this library.

closed 2021-04-22 by khoih-prog

Szum123321, 2021-04-23:
The library is included in include/I2CEEPROM.h file

khoih-prog, 2021-04-23:
You have to modify your code as follows

1. main.cpp

```
#include <Arduino.h>
#include "TimerInterrupt_Generic.h"
#include "I2CEEPROM.h"

I2CEEPROM mem;

void setup() 
{
}

void loop() 
{
}
```

---

2. I2CEEPROM.h

```
#include <Arduino.h>
#include <Wire.h>
#include <math.h>

#ifdef __STM32__
  STM32Timer DEVICE_TICK_TIMER(TIM2);
#elif defined(__TEENSY__)
  TeensyTimer DEVICE_TICK_TIMER(TEENSY_TIMER_3);
#endif

#ifndef I2CEEPROM_H
#define I2CEEPROM_H

//#define __STM32__
#define __TEENSY__

class I2CEEPROM {};

#endif
```

---

3. I2CEEPROM.cpp

```
#include "I2CEEPROM.h"

/*library code here*/
```
---

Compile will be OK as tested on Arduino IDE, unless your PIO settings /configuration are wrong

[image]

khoih-prog, 2021-04-23:
You have to modify like this to use new C++ style hpp

1. main.cpp or Teensy_minimal.ino

```
#include <Arduino.h>
#include "I2CEEPROM.h"

I2CEEPROM mem;

void setup() 
{
}

void loop() 
{
}
```

---

2. I2CEEPROM.h

```
#ifndef I2CEEPROM_H
#define I2CEEPROM_H

#include <Arduino.h>
#include <Wire.h>
#include <math.h>
#include "TimerInterrupt_Generic.h"

//#define __STM32__
#define __TEENSY__

class I2CEEPROM {};

#include "I2CEEPROM_Impl.hpp"

#endif
```

---

3. I2CEEPROM_Impl.hpp

```
#ifndef I2CEEPROM_IMPL_HPP
#define I2CEEPROM_IMPL_HPP

#ifdef __STM32__
  STM32Timer DEVICE_TICK_TIMER(TIM2);
#elif defined(__TEENSY__)
  TeensyTimer DEVICE_TICK_TIMER(TEENSY_TIMER_3);
#endif

/*library code here*/

#endif    // I2CEEPROM_IMPL_HPP
```
---

Compile is OK on Arduino IDE, and should be OK on PIO, unless you have wrong settings / configurations

[image]