khoih-prog/TimerInterrupt_Generic #4: Doesn't compile for nrf52 boards with Arduino IDE

shanyechungu, 2021-08-11:
Arduino: 1.8.15 (Mac OS X), Board: "Adafruit Feather nRF52840 Sense, S140 6.1.1, Level 0 (Release)"

/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: sketch/TimerInterruptLEDDemo.ino.cpp.o: in function `doingSomething1()':
/var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:91: undefined reference to `Serial'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: sketch/TimerInterruptLEDDemo.ino.cpp.o: in function `doingSomething2()':
/var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:97: undefined reference to `Serial'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: sketch/TimerInterruptLEDDemo.ino.cpp.o: in function `doingSomething3()':
/var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:102: undefined reference to `Serial'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: sketch/TimerInterruptLEDDemo.ino.cpp.o: in function `setup':
/var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:107: undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:108: undefined reference to `Adafruit_USBD_CDC::operator bool()'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/4p/g4vjjkm91599mvnrhgf992kw0000gp/T/arduino_modified_sketch_899459/TimerInterruptLEDDemo.ino:136: undefined reference to `Serial'
/Users/NNL/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: sketch/TimerInterruptLEDDemo.ino.cpp.o: in function `ISR_Timer::findFirstFreeSlot()':
/Users/NNL/Documents/Arduino/libraries/TimerInterrupt_Generic/src/ISR_Timer-Impl_Generic.h:158: undefined reference to `Serial'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit Feather nRF52840 Sense.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

khoih-prog, 2021-08-12:
Same as [**Can't compile in Arduino IDE with Feather Sense board** #1](https://github.com/khoih-prog/NRF52_TimerInterrupt/issues/1)

You have to use **nRF52 core v0.21.0 or earlier.**

closed 2021-08-12 by khoih-prog

khoih-prog, 2021-08-13:
FYI, the new release v1.7.0 has just been released to fix the issue. Tested OK with latest nRF52 core v0.24.0
Please try and report if there is any error.

---

### Releases v1.7.0

1. Add support to Adafruit nRF52 core v0.22.0+
2. Add support to nRF52 Raytac MDBT50Q_RX Dongle

**Warnings for Releases v1.6.0-** : Use [`Adafruit nRF52 v0.22.0+`](https://github.com/adafruit/Adafruit_nRF52_Arduino/releases/tag/0.22.0) will create **compiler errors**. If so, please use the [`Adafruit nRF52 v0.21.0`](https://github.com/adafruit/Adafruit_nRF52_Arduino/releases/tag/0.21.0). It's advisable to update to Releases v1.7.0+