khoih-prog/TimerInterrupt_Generic #5: PlatformIO and Heltec WiFi LoRa V2

MicSchi, 2021-12-23:
### Describe the bug

Seems i cannot make it compile because it cannot find some function. Googling for it does not reveal any help. I have no idea how to overcome this issue.

- timer_isr_callback_add 
- TIMER_INTR_T0
- TIMER_INTR_T1

### Steps to Reproduce

Visual Studio Code 1.63.2 + PlatformIO Core 5.2.4. + ESP32 + #include "TimerInterrupt_Generic.h"

### Expected behavior

Compile successfully

### Actual behavior
Terminal output when compiling

Processing heltec_wifi_lora_32_V2 (platform: espressif32; board: heltec_wifi_lora_32_V2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32_V2.html
PLATFORM: Espressif 32 (3.4.0) > Heltec WiFi LoRa 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Heltec ESP32 Dev-Boards> 1.1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <TimerInterrupt_Generic> 1.8.0
|-- <DHT sensor library for ESPx> 1.18
|-- <Freenove WS2812 Lib for ESP32> 1.0.5
|-- <SPI> 1.0
|-- <U8g2> 2.31.2
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <Wire> 1.0.1
|-- <Ai Esp32 Rotary Encoder> 1.4
Building in release mode
Compiling .pio\build\heltec_wifi_lora_32_V2\src\WeckerDeluxe.cpp.o
In file included from .pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/TimerInterrupt_Generic.h:221:0,
                 from src\WeckerDeluxe.cpp:38:
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h:90:22: warning: "/*" within comment [-Wcomment]
   TIMER_GROUP_0 = 0, /*!<Hw timer group 0
                      ^
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h:91:22: warning: "/*" within comment [-Wcomment]
   TIMER_GROUP_1 = 1, /*!<Hw timer group 1
                      ^
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h:100:16: warning: "/*" within comment [-Wcomment]
   TIMER_0 = 0, /*!<Select timer0 of GROUPx
                ^
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h:101:16: warning: "/*" within comment [-Wcomment]
   TIMER_1 = 1, /*!<Select timer1 of GROUPx
                ^
Compiling .pio\build\heltec_wifi_lora_32_V2\libda1\U8g2\clib\u8g2_message.c.o
Compiling .pio\build\heltec_wifi_lora_32_V2\libda1\U8g2\clib\u8g2_polygon.c.o
Compiling .pio\build\heltec_wifi_lora_32_V2\libda1\U8g2\clib\u8g2_selection_list.c.o
In file included from .pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/TimerInterrupt_Generic.h:221:0,
                 from src\WeckerDeluxe.cpp:38:
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h: In member function 'bool ESP32TimerInterrupt::setFrequency(float, esp32_timer_callback)':
.pio\libdeps\heltec_wifi_lora_32_V2\TimerInterrupt_Generic\src/ESP32TimerInterrupt_Generic.h:307:100: error: 'timer_isr_callback_add' was not declared in this scope     
         timer_isr_callback_add(_timerGroup, _timerIndex, _callback, (void *) (uint32_t) _timerNo, 0);
[... 81 lines cut ...]
#include "heltec.h"   // WiFi LoRa 32 (V2)

// The Timer for the different things
// These define's must be placed at the beginning before #include "ESP32_New_TimerInterrupt.h"
// _TIMERINTERRUPT_LOGLEVEL_ from 0 to 4
#define _TIMERINTERRUPT_LOGLEVEL_     0
#include "TimerInterrupt_Generic.h"

```

### Additional context

; platformio.ini
[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
board_build.mcu = esp32
board_build.f_cpu = 240000000L
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
upload_port = COM8
lib_compat_mode = strict
lib_deps = 
	heltecautomation/Heltec ESP32 Dev-Boards@^1.1.0
	khoih-prog/TimerInterrupt_Generic@^1.8.0

MicSchi, 2021-12-23:
Just tested with Arduino IDE 1.8.16 and Example -> TimerInterrupt_Generic -> ESP32 -> Argument_None resulting in the same error messages.

khoih-prog, 2021-12-23:
> framework-arduinoespressif32 3.10006.210326 (1.0.6)

Please read [**Prerequisites**](https://github.com/khoih-prog/TimerInterrupt_Generic#prerequisites), and **update ESP32 core to v2.0.1+** to avoid wasting everybody's time

[image]

closed 2021-12-23 by khoih-prog

MicSchi, 2021-12-24:
Thank you for your fast answer, it is very much apprechiated! It help me to get further with the topic after beeing stuck for nearly a day. Didn't intent to waste anybodys time. Just new to ESP32 topic and overloaded by the massive amount of information available.

Maybe interesting for others: It seems that the Heltec board is tight to the v1.0.6 at the moment. Installing esp32 V2.0.2 does not improve the situation. Update wasn't possible because wasn't installed before. 
I assume espressif v1.0.6 is integral part of the Heltec board package and to utilize ESP v2.0.2 package I would need to switch to one of its supported boards. 
[image]
It seems the Heltec package needs changes to incorporate espressif v2.0.1+. Therefore I added a comment to an existing comparable issue there: [Heltec-Aaron-Lee/WiFi_Kit_series](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/150)