khoih-prog/TimerInterrupt_Generic #8: Support for DX CORE CPU and MightyCORE CPU possible?

LaurentR59, 2022-08-22:
### Is your feature request related to a problem? Please describe.

As alter part AVR DA DB DD and modern ATtiny CPU are interesting to use
DX CORE is closed to MEGACOREX.

### Describe the solution you'd like

Support generic timer mode like ATMEGA4809 CPU for new CPU (ATTINY series 0 1 () 2 and AVR DA DB (DD?)

### Describe alternatives you've considered

none existing or adjustment for supporting new parts

### Additional context

Put a look at DX CORE lib and mithtycore from Spence Konde.

Thanks in advan,ce t consider benefits of enlargment copatibility CPU supported.

Regards
Laurent

khoih-prog, 2022-08-22:
Hi @LaurentR59 

Thanks for the info and request. I'll have a look and try when having time, which is hard for me to reserve and find. So I can't promise now.

I already install the [**DxCore**](https://github.com/SpenceKonde/DxCore) and **megaTiny** core

BTW, how and where can I find and buy the eval boards / products, **working with Arduino IDE**, to test ? I'd appreciate if you or anybody can donate the boards / system to test.

Regards,

LaurentR59, 2022-08-22:
Hello

Probably best test cards for tests are respectively the curiosity nano XXX boards.

Cards are supported by libraries and IDE tools. (like platformio or arduino IDE)

For DX CORE AVR DA DB, the curiosity nano avr128da48  and 128db48 from microchip are available for exemple at MOUSER. ( but on some others dealers too or directly from Microchip)

Curiosity nano 128DA48
https://www.mouser.fr/ProductDetail/Microchip-Technology-Atmel/EV35L43A?qs=sPbYRqrBIVm2MSslyKFSbQ%3D%3D

Curiosity nano 128DB48
https://www.mouser.fr/ProductDetail/Microchip-Technology-Atmel/DM164151?qs=vmHwEFxEFR8q2uCPVI5P2g%3D%3D

About Attiny probably it should be this one with ATTINY3217 CPU:
https://www.mouser.fr/ProductDetail/Microchip-Technology-Atmel/EV50J96A?qs=OlC7AqGiEDlhXXlje7rJaw%3D%3D

I have only one of them the curisoity nano128DA48  and made a breadbord with ATTINY1606 (but require UDPI programmer) so may it help?

Thanks for your time on such futur lib update.
Laurent

khoih-prog, 2022-08-22:
Thanks for the info.

I've already ordered those boards and will have a look when they arrive

#### Curiosity nano 128DA48

[image]

#### Curiosity nano 128DB48

[image]

#### ATtiny3217 Curiosity Nano Evaluation Kit

[image]

LaurentR59, 2022-08-22:
I made a error on mightycore label It  should be more megaTinyCore from Spece Konde too like DX CORE.

Sorry for mistake anyway.
Should be a good step to get them!

Laurent

khoih-prog, 2022-08-22:
Too late.

I almost finish creating the first **ATmega_TimerInterrupt** library for  [**MightyCore**](https://github.com/MCUdude/MightyCore).
Will publish within hours.

If you have board, please test and report as I'm **dry-coding**, without any `ATmega???4` board.

---

### Releases v1.0.0

1. Initial coding for AVR`ATmega164`, `ATmega324`, `ATmega644` and `ATmega1284` using [**MightyCore**](https://github.com/MCUdude/MightyCore)

khoih-prog, 2022-08-23:
Hi @LaurentR59 

The new [**ATmega_TimerInterrupt releases v1.0.0**](https://github.com/khoih-prog/ATmega_TimerInterrupt/releases/tag/v1.0.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/ATmega_TimerInterrupt#contributions-and-thanks)

Please test and report any bug in [**Issues**](https://github.com/khoih-prog/ATmega_TimerInterrupt/issues)

Best Regards,

---

### Releases v1.0.0

1. Initial coding for AVR **ATmega164(A/P), ATmega324(A/P/PA/PB), ATmega644(A/P), ATmega1284(P)** using [**MightyCore**](https://github.com/MCUdude/MightyCore)

khoih-prog, 2022-08-23:
Hi @LaurentR59 

The new [**ATmega_Slow_PWM releases v1.0.0**](https://github.com/khoih-prog/ATmega_Slow_PWM/releases/tag/v1.0.0) has just been published.

Please test and report any bug in [**Issues**](https://github.com/khoih-prog/ATmega_Slow_PWM/issues)

Best Regards,

---

### Initial Releases v1.0.0

1. Initial coding for AVR **ATmega164(A/P), ATmega324(A/P/PA/PB), ATmega644(A/P), ATmega1284(P)** using [MightyCore](https://github.com/MCUdude/MightyCore)
2. The hybrid I**SR-based PWM channels** can generate from very low (much less than 1Hz) to highest PWM frequencies up to 500Hz with acceptable accuracy.

khoih-prog, 2022-08-26:
1. [Dx_TimerInterrupt library](https://github.com/khoih-prog/Dx_TimerInterrupt)

---

## Changelog

### Release v1.1.2

1. Using `Serial3` for debugging with `Curiosity Nano AVRDB`, and `Serial1` for debugging with `Curiosity Nano AVRDA`

### Release v1.1.1

1. Using Serial1 instead of Serial for debugging with Curiosity Nano AVRDA/AVRDB
2. Update `README.md`

### Release v1.1.0

1. Improve and customize examples for `Curiosity Nano AVRDA/AVRDB` boards to use on-board LED and SW
2. Add notes `howto upload by drag-and-drop` to `CURIOSITY` virtual drive

### Initial Release v1.0.0

1. Intial release to support Arduino **AVRDx-based boards (AVR128Dx, AVR64Dx, AVR32Dx, etc.) using DxCore**

---
---

2. [Dx_Slow_PWM library](https://github.com/khoih-prog/Dx_Slow_PWM)

---

## Changelog

### Release v1.0.2

1. Minor cosmetic fix

### Release v1.0.1

1. Make `MAX_NUMBER_CHANNELS` configurable to max **64 PWM channels**
3. Remove debug codes possibly causing hang
4. Improve debug to use `Serialx` port automatically according to boards

### Initial Release v1.0.0

1. Initial release to support Arduino **AVRDx-based boards (AVR128Dx, AVR64Dx, AVR32Dx, etc.) using DxCore**

khoih-prog, 2022-08-26:
I'm closing the issue now as done.

closed 2022-08-26 by khoih-prog